Pfalra/FFFilamentrecycling #5: Unidentified issue in temperature calculation

Pfalra, 2022-02-27:
The temperature calculation sometimes jumps up and down by 10 degC within 500 ms as soon as you touch the thermistor.
After releasing the thermistor the temperature jumps from around 34 deg to 19 deg within 500 ms.
This behaviour indicates a somewhat broken calculation method that needs to be looked into.

g3gg0, 2022-02-27:
What do the raw ADC values tell?

Pfalra, 2022-02-28:
Just checked the raw values. Seems like temperature calculation is messed up. In the near future we are going to use Steinhart-Hart method anyway. So this will no longer be an issue.

Pfalra, 2022-03-03:
Implemented Steinhart-Hart method. Hoever now calculation does not reach more than 28 degrees Celcius. Seems like coefficients are not calculated properly.
Solution could be accepting precalculated coefficients.

Pfalra, 2022-03-06:
Just more or less fixed the temperature calculation. Sometimes it is still a bit jumpy, but that may be due to vcc not being properly stable. I am using caps to filter a bit though. I'll take a look at it with my oscilloscope. However for our purposes jumps of +- 5°C should be acceptable. It may affect viscosity but therefore we have the pid control to mitigate it.

closed 2022-03-06 by Pfalra